elodin-sys/elodin #11: AI generated documentation for Nox

nestordemeure, 2024-03-27:
Hi!

Would you be interested in a PR adding AI-generated documentation to the types, traits, and functions in Nox? It would be manually checked by a human (me) who has knowledge of Jax and Rust but not this particular codebase.

I have run some quick tests and the output seem serviceable, definitely not perfect but a good starting point to be tweaked by people familiar with the codebase before merging.

akhilles, 2024-03-28:
Hey @nestordemeure, we're certainly open to it but it depends heavily on the quality of the generated comments. Could we start with a small submodule and then review it?

My general preference for code comments is to not be overly verbose wrt. information that can easily be inferred from the function signature. Comments that explain complex/unintuitive behavior are the most welcome.

nestordemeure, 2024-03-28:
Do you have a submodule you would recommend? I could then make a PR to have comments to discuss.

sphw, 2024-03-28:
> Do you have a submodule you would recommend? I could then make a PR to have comments to discuss.

Could you try on the `tensor` module?

nestordemeure, 2024-03-28:
Will do and keep you posted once the PR is up :)

nestordemeure, 2024-03-31:
I think the `tensor.rs` trial was a success. My suggestion would be to keep going one file at a time (that way you have time to review the comments without being overwhelmed). In which case, in which order would you like me to cover the Nox files?

akhilles, 2024-04-02:
1 PR for the rest of the module is fine, and somewhat preferable. Reviewing it isn't a concern.

nestordemeure, 2024-05-01:
I have added a PR covering the full Nox library. That was a *lot* of comments and will definitely take some time to review but it looks like an acceptable basis to document the code.

akhilles, 2024-05-01:
> I have added a PR covering the full Nox library. That was a _lot_ of comments and will definitely take some time to review but it looks like an acceptable basis to document the code.

Thanks! Will try review it today or tomorrow.

closed 2024-08-21 by akhilles