elodin-sys/elodin #20: Docs.rs fails to build the Nox documentation

nestordemeure, 2024-11-27:
[Docs.rs fails to build the Nox documentation](https://docs.rs/crate/nox/latest), meaning that Crates.io cannot display a documentation link on the [crate's page](https://crates.io/crates/nox/0.4.0).[^readme]

[^readme]: On a related note, the crate does not have a README to display on its main crates.io page to orient newcomers as to what it does and where to find further information.

kaf-lamed-beyt, 2024-12-08:
Hi @nestordemeure , do you think this can be categorized as a "good first issue"? I recently started experimenting with Rust and I'd love to attempt the issue if you can point me in the direction. Thanks!

nestordemeure, 2024-12-08:
Hi @kaf-lamed-beyt, I believe it would be! It would be rust adjacent since it has to do with writing documentations and the way it is displayed by the repositories, [the above link](https://docs.rs/crate/nox/latest) point at ways to try and debug it.

However, you will need to reproduce the building error locally before you can try and fix it.

kaf-lamed-beyt, 2024-12-08:
Alight! Sure thing. I'll take a look at the link you shared, and I'll also try reproducing the error. 

I'll keep you posted.

sphw, 2024-12-08:
Thanks for filing this issue @nestordemeure. @kaf-lamed-beyt I'd be happy for you to talk a look at fixing this, but I wouldn't necessarily classify it as a good first issue. 

The core problem is that `noxla`, our wrapper around XLA, requires the XLA's headers to compile. Normally we download XLA on each build, but docs.rs disables network access while building documentation. There are a few fixes that we could implement, but the easiest is probably to vendor XLA's header files into `noxla`. You would then need to modify build.rs to skip downloading the XLA extension when built on docs.rs. You can find docs on how to do that here: https://docs.rs/about/builds

kaf-lamed-beyt, 2024-12-08:
Oh! thank you for this feedback @sphw

So, am I looking at the build.rs file in `nox` or `noxla` lib?

sphw, 2024-12-08:
> So, am I looking at the build.rs file in `nox` or `noxla` lib?

`noxla`

kaf-lamed-beyt, 2024-12-08:
super! I'll let you know if i run into any blockers!

kaf-lamed-beyt, 2024-12-08:
@sphw, I tried doing `cargo run` in the root, but I got this error 

```shell
cargo run
    Updating crates.io index
    Updating git repository `https://github.com/elodin-sys/big_space.git`
error: failed to select a version for `faer`.
    ... required by package `nox v0.1.0 
    ... which satisfies path dependency `nox` of package `impeller v0.1.0 
    ... which satisfies path dependency `impeller` of package `nox-py v0.1.0 `
versions that meet the requirements `^0.19.1` are: 0.19.4, 0.19.3, 0.19.2, 0.19.1

the package `nox` depends on `faer`, with features: `cholesky` but `faer` does not have these features.

failed to select a version for `faer` which could resolve this conflict
```

sphw, 2024-12-09:
@kaf-lamed-beyt Try pulling and rebuilding. We just pushed changes that should fix this

kaf-lamed-beyt, 2024-12-11:
@sphw, I got the same error still

sphw, 2024-12-11:
> @sphw, I got the same error still

This PR should fix it https://github.com/elodin-sys/elodin/pull/21 . You can try off the branch – we will merge it soon.

kaf-lamed-beyt, 2024-12-12:
@sphw, this is what i got: 

```shell
error: failed to run custom build command for `nox v0.1.0 (/oss/elodin/libs/nox)`

Caused by:
  process didn't exit successfully: `/oss/elodin/target/debug/build/nox-38fe8fc81b574a5d/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at libs/nox/build.rs:48:50:
  called `Result::unwrap()` on an `Err` value: Os { code: 2, kind: NotFound, message: "No such file or directory" }
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
warning: noxla@0.1.0: src/client.rs:37: warning: ignoring '#pragma clang diagnostic' [-Wunknown-pragmas]
warning: noxla@0.1.0:    37 |     #pragma clang diagnostic ignored "-Wundefined-var-template"
warning: noxla@0.1.0:       |
warning: noxla@0.1.0: In file included from /usr/include/c++/11/backward/strstream:50,
warning: noxla@0.1.0:                  from src/hlo_module.rs:6:
warning: noxla@0.1.0: /usr/include/c++/11/backward/backward_warning.h:32:2: warning: #warning This file includes at least one deprecated or antiquated header which may be removed without further notice at a future date. Please use a non-deprecated interface with equivalent functionality instead. For a listing of replacement headers and interfaces, consult the file backward_warning.h. To disable this warning use -Wno-deprecated. [-Wcpp]
warning: noxla@0.1.0:    32 | #warning \
warning: noxla@0.1.0:       |  ^~~~~~~
```

Although, I suspect this may be due to my internet connection. I'll keep trying though.

sphw, 2024-12-12:
> Although, I suspect this may be due to my internet connection. I'll keep trying though.

Try building from within the libs/nox directory. That error happens when you are missing a Python installation with shared libraries – it should only happen when building `nox-py`. If you build within the `libs/nox` directory you should bypass the error. If you want to build `nox-py` the easiest way is to make a [uv](https://docs.astral.sh/uv/) venv.