elodin-sys/elodin #138: Add menu bar for Schematic features

zxq82lm, 2025-09-18:
Menu items to include:
- Save KDL: export the current schematic to a .kdl file.
- Open KDL: load a schematic from an existing .kdl file.
- Reset KDL / New Project: clear the current schematic to start a new simulation without quitting and reopening the application.

zxq82lm, 2025-09-24:
> I was first to suggest a menu bar, but I'm not certain we need a menu bar yet. Perhaps we just need some kind of on screen help button that people can press and it will tell them about command palette and its hot key.

I can repurpose this ticket into simply adding a help button that explains the command palette.
Or we can just close the issue.
What’s your preference?

x46085, 2025-09-24:
> I can repurpose this ticket into simply adding a help button that explains the command palette. Or we can just close the issue. What’s your preference?

Could even be a tiny unobtrusive button that opens the command palette? (with a small note of the hot-key for next time?)

Btw if you ever need to bounce UX ideas off an expert, @Teegurks is available to chat with (tom@elodin.systems)