elodin-sys/elodin #143: Create READMEs for each example in libs/nox-py/examples

zxq82lm, 2025-09-18:
Proposed structure:
```
examples/
├── cube-sat/
│   ├── cube-sat.py
│   └── cube-sat-pysim.py
│   └── README.md
├── rocket/
│   ├── rocket.py
│   └── README.md
├── three-body/
│   ├── three-body.py
│   └── README.md
```

Each README can briefly describe:
- what the example does,
- required dependencies,
- how to run it

x46085, 2025-09-18:
You'll need to look at the use of the Elodin CLI in order to fully understand this change. Currently the CLI allows for creating "templates" from the examples, such as the command `elodin create --template ball` and `elodin create --template rocket`. Definitely do the associated ticket https://github.com/elodin-sys/elodin/issues/83 either before or together with this change, and make sure the CLI is still working.

anderspitman, 2025-10-02:
Do we want to consolidate the examples into `./libs/examples` as described here, or into `./examples`, which is how I read #83, and seems a bit more intuitive to me?

x46085, 2025-10-02:
> Do we want to consolidate the examples into `./libs/examples` as described here, or into `./examples`, which is how I read [#83](https://github.com/elodin-sys/elodin/issues/83), and seems a bit more intuitive to me?

Originally I wanted top level, but in doing some development with nox-py, have those examples relative to the locally built python wheel works pretty well, and would become slightly more arduous having to point a few folders down. Give that a try and see how it it feels, if it’s fine and nox-py Readme is updated and seems intuitive, then yes top level is preferred.

Note that moving those locations will also break the cli template create command, and you’ll have to fix it or deprecate the command (preferred, but probably should be its own PR with all the code / docs / Readme updates required)

anderspitman, 2025-10-06:
> [@x46085](https://github.com/x46085) to clarify, you're saying you would prefer the `create` command to be deprecated regardless of whether it can be made to work with examples in a different directory?

We discussed this on a call and decided to go ahead and deprecate the `create` command.